JYU-IBA/potku #3: Move tof.in to request folder

jussiks, 2020-03-02:
tof.in contains request specific settings file that are used as parameters for tof_list. It is currently located in the external/Potku-bin folder, but a more logical location would be the folder where the request is saved ([request name].potku). Also, this file gets rewritten during the runtime of the program so it makes little sense to have it in version control.

jaakkojulin, 2020-06-11:
Commit edec97eb90c5bd7335201fec3c0869dd5006d1bb contains a fix for tof_list that removes the hard coded tof.in file. Now this file should be given as the first argument, before the cutfiles.

There is another fix: the old code had very bad filename extension parsing and wouldn't work if you had commas (.) somewhere in the path. I think this is the reason why Potku copies cutfiles to temporary directory before giving them to tof_list?

Note that the modifications to tof_list rely on [_splitpath](https://docs.microsoft.com/en-us/cpp/c-runtime-library/reference/splitpath-wsplitpath?view=vs-2019) on Windows and this code hasn't been tested on Windows yet. The unix solution uses basename() and should work on Linux and macOS.

Changes to Python code are not implemented in this branch, so therefore no PR.

jussiks, 2020-06-11:
I'm running into character encoding issues on Windows after compiling the new version of tof_list:
```
tof_list.exe: config from tof.in, 1 cut files to process.
argv[0] = "tof_list.exe"
argv[1] = "tof.in"
argv[2] = "m1.Si.ERD.0.cut"

file 0 is "m1.Si.ERD.0.cut"
ZZ=0 (mass number), symbol[0]=O�|u{
Trying to find mass for "O�|u{" (mass number A is 0)
Could not find element O�|u{
```
File paths omitted for brevity.

Happens when I run it in Powershell or from Potku's Python code.

jaakkojulin, 2020-06-11:
Probably not an encoding issue, just the file extension code failing and producing garbled output due to unsafe pointer code.  I think my misunderstanding was that _splitpath splits the LAST extension, i.e. ".cut" and not ".Si.ERD.0.cut". I might have fixed it in commit 1424b6e0c3b1b854979de4b7e3663b78db17541a but I probably need to test it on Windows. Writing perfect C without compiling is hard, even for me :)

jussiks, 2020-06-11:
With a [smallish bug fix :)](https://github.com/JYU-IBA/potku/commit/da9f82a0c77afcd5a9fd68e59f2ca408fe01ee27) tof_list now seems to be working on Windows too.

Issue branch now seems ready to be merged and this issue can also be closed.

PR #43 (2020-06-11): Issue 3 tofin
This partially fixes #3. tof.in is written to `<request>/<sample>/<measurement>/tof_in` folder. However, the file still needs to be copied to bin dir as tof_list has to have it in its working directory and also tof_list has to be able to masses.dat file with a relative path.

This pull request also features lots of changes to how Meausurement objects are serialised and deserialised. .info file and other settings are no longer deleted when user ticks "Use request settings". The logic is now same as with Simulation objects.
closed 2020-06-12 by jaakkojulin