JYU-IBA/potku #4: Should temporary files be written to somewhere other than /tmp?

jussiks, 2020-03-02:
Currently Potku uses /tmp folder or ~\AppData\Local\Temp to store temporary files. However, to ensure read/write permissions, it might be a good idea to place them in the request instead. The files could be written to a "hidden" folder called [request name].potku/.tmp/.

jaakkojulin, 2020-03-05:
Temporary files should be placed in tempfile.gettempdir() like they currently are. This doesn't simply return a fixed path, but actually tries to make sure there are write permissions. Current working directory is the last resort.

However, the feature to save the temporary files under the request might be useful, especially for the "tof.in" file.

jussiks, 2020-03-10:
Yeah, current usage of /tmp seems good enough for the desktop version. There is a possible problem with multiple processes reading/writing the same files at the same time, but that is a bigger concern for the upcoming server version (and should be its own issue).

closed 2020-03-10 by jussiks