JYU-IBA/potku #9: Making depth profiles with one ore more scattered element (RBS) doesn't work.

jaakkojulin, 2020-03-05:
Using commit d4e20e577be1e11edc38700dee7853146c94146e (development branch) with OSX and Python 3.7.6. Making depth profiles with scattered element (RBS) doesn't work.

I am trying to use two heavy elements Xx and Yy with beam 123Bb. I get cut files "measurement.123Bb.RBS_Xx.0.cut" and "measurement.123Rb.RBS_Yy.0.cut" and these look ok. If I choose them both for depth profiling it looks like tof_list fails somehow and I don't get all the depth files that I am supposed to get (many elements are missing).

If I choose only one (for example Xx), I get all the depth files, including depth.Xx. However the element "Xx" is not plotted by Potku. 

So there are at least two bugs, one in tof_list and one in the plots or depth profile handling.

jaakkojulin, 2020-03-05:
I'll try to fix the bug in tof_list...

jussiks, 2020-03-05:
If I try to run tof_list on my Fedora machine, I get following output:

```
$ ./tof_list /tmp/m1.2H.ERD.0.cut 
Directory: ~/projektit/potku/gradu_testi.potku/Default/Detector/Efficiency_files/Used_efficiencies
Efficiency file: ~/projektit/potku/gradu_testi.potku/Default/Detector/Efficiency_files/Used_efficiencies/2H.eff
0.000000e+00 0.000000e+00    1.11071   1   2.0141 ERD  1.000   644
0.000000e+00 0.000000e+00    1.07445   1   2.0141 ERD  1.000   764
[...]
0.000000e+00 0.000000e+00    1.08875   1   2.0141 ERD  1.000 113722
free(): double free detected in tcache 2
Aborted (core dumped)
```

When I try to generate energy spectra, nothing gets drawn as the tof_output is either empty or an index error happens when trying to parse the output.

tof_list function in the modules.general_functions module also contains a bug. It tries to return tof_output in the finally block, even though tof_output does not exist if there has been an exception. I can fix that one.

jaakkojulin, 2020-03-05:
The main bug was fixed by setting  herp_isotope=0 for each cut file. The isotope number wasn't reset for the second file and so the behaviour was quite unexpected. Additionally I found issues with stopping assignments when a specific isotope was selected (unrelated to this bug).

jaakkojulin, 2020-03-05:
The emax[i] calculation wasn't also correct for scattered beam, so I made a simplification to this code by assuming the emax is 1.2 times the incident beam energy. Should work.

jussiks, 2020-03-05:
I merged my Python fix to the issue-9 branch. I still could not get tof_list to work on Fedora though. I'll test it on Mac and Windows (at least Windows version has worked for me previously).

jussiks, 2020-03-05:
Just to clarify, does this energy spectra look correct:
[image]

Previous version of the tof_list actually failed to generate spectrum for Rhenium so this is the first time I'm seeing it. Also depth profile looks funny when you include Rhenium...

Anyway, this now also works on Mac too, so we can probably merge the branch to development.

jaakkojulin, 2020-03-05:
The Rhenium is actually not Rhenium, it's probably just a "reference" (Re) selection of silicon to be used when looking at losses.

closed 2020-03-05 by jaakkojulin