JYU-IBA/potku #26: Fix starting a simulation with no Target layers

jussiks, 2020-04-03:
If the Target has no layers, MCERD halts immediately because the random number generator generates a -NaN. The console output looks as follows:
```
0.000000 inf -nan 1 1
Fatal error: Illegal value from random number generator
```
Python code does not currently handle this situation. Simulation remains in a perpetual "Pre-sim mode", not giving the user any indication that something has gone wrong.

This issue may be solved by fixing the RNG (which is its own issue), but also Potku should be able to better handle situations where simulation stops abruptly.

mlaitin, 2020-04-03:
Shortly as i wrote to Juhani S already, target layers need to be set by default first, before user can be passed to modify recoil distributions.

Target defines stopping/ion transport conditions which are fairly independent from the recoil distribution, this is "the key, and invention" of the MCERD process

    mikko

--
Mikko Laitinen, Researcher
University of Jyväskylä
Department of Physics
________________________________________________________
Postal address: University of Jyväskylä,
Department of Physics, P.O. Box 35,
40014, Finland
Street address: Survontie 9 D, Ylistönrinne, Jyväskylä

Office: +358 40 8054 113
Personal: +358 400 99 48 36
Telefax: +358 14 617 411
E-mail address: mikko.i.laitinen@jyu.fi

What was necessary, was done by yesterday;
We're currently working on the impossible;
For miracles, we ask for a 24 hours notice ... :)

jussiks, 2020-04-03:
I think this bug can only appear if the user defines the target layers first, creates some recoil elements and then deletes the Target layers.

So basically the software behaves inconsistently when the user is using it "wrong".  Not a major or common bug in itself but something the software should be prepared for.

Solution could be to always prevent the user from starting simulations or editing recoil elements when target layers do not exist, which makes this issue related to #25.

jussiks, 2020-05-12:
Potku now shows an error message if this happens (although the error message just says 'MCERD stopped with an error code 10' which is not very helpful for the user). This will do for now, especially with the new version of MCERD coming, so I'll close this one.

closed 2020-05-12 by jussiks