JYU-IBA/potku #58: test_imports fails on Developer CMD

tutapitk, 2020-07-02:
For some reason, tests/integration/test_imports.py fails on Developer Command Prompt while packaging Potku. I tried this on the current master branch (0f6d4ec7abfac47f2001298b37ada35256a9b64f).

Last week, I could create the bundle with no issues (82d21f786549063899ca2bf67f55379d0d49b756, I think).

Curiously, packaging Potku with `pyinstaller potku.spec` still works.

Used commands:
```
pipenv shell
python -m unittest discover
```

Or (on Developer Command Prompt or Developer Powershell):
```
pipenv run create_bundle.bat
```

Tests succeed on:
- PyCharm (Run 'Unittests in test')
- Command Prompt
- Developer PowerShell for VS 2019 (and probably other versions)

test_imports fails on:
- Developer Command Prompt for VS 2015
- Developer Command Prompt for VS 2017
- Developer Command Prompt for VS 2019

`pyinstaller spec.potku` works on (at least):
- Developer Command Prompt for VS 2017

I'm have no idea why this happens. Maybe my ancient installation of Windows is acting up again. I'll try this on my other Windows machine at some point.

----

Error message:

> ERROR: test_importing_modules (tests.integration.test_imports.TestImports)
> Tests that modules are not imported twice when they are
> ----------------------------------------------------------------------
> Traceback (most recent call last):
>   File "d:\kurssit\tietotekniikka\potku_kesa\potku\paketointitesti\potku\tests\utils.py", line 79, in wrapper
>     func(*args, **kwargs)
>   File "d:\kurssit\tietotekniikka\potku_kesa\potku\paketointitesti\potku\tests\integration\test_imports.py", line 61, in test_importing_modules
>     import potku.modules.masses as m1
> ModuleNotFoundError: No module named 'potku.modules'; 'potku' is not a package

jussiks, 2020-07-03:
That test was written with web version in mind. There were some unwanted side effects like modules being imported twice and broken intellisense when importing potku modules in a working directory that was up one folder from Potku's root. The test is not particularly important in the context of the GUI application as imports have always worked as intended for that version. Failure should not affect pyinstaller's ability to package the source code.

You can remove the test as it was more about me figuring out the Python import system rather than testing Potku itself. If you want to troubleshoot the issue more, you could add some print statements (sys.path, current working directory etc) and run the tests with "python -m unittest discover" from developer prompt.

jussiks, 2021-02-01:
I had the same issue after upgrading pipenv to version 2020.11.15. The issue seems to happen because the root directory does not get removed from sys.path. I do not really see this as worth fixing, so it is just easier to remove the test altogether.

PR #116 (2021-02-01): Remove test_imports.py (#58)
- test_imports was not working properly. When run in Developer Prompt,
  the root directory was not being removed from sys.path, causing the
  test to fail. It is easier to just remove the test rather than fix
  it. Closes #58.
- Change the import in profile.py to relative. This is the style that
  has been used in other modules in the modules package.
commit 5e4220a8 (2021-02-01): Remove test_imports.py (#58)

- test_imports was not working properly. When run in Developer Prompt,
  the root directory was not being removed from sys.path, causing the
  test to fail. It is easier to just remove the test rather than fix
  it. Closes #58.
- Change the import in profile.py to relative. This is the style that
  has been used in other modules in the modules package.
closed 2021-02-01 by jussiks