JYU-IBA/potku #76: Depth Profile does not work

SjoerdB93, 2020-08-26:
Hello,

I am not completely sure if this is the right place, but depth profile calculations do not seem to work in Linux (Fedora). Everything else seems to work fine, but as soon as I want to calculate my depth profile the window remains gray and I get the following error in the built-in log:

`2020-08-26 16:14:04 - ERROR - Could not create Depth Profile graph: [Errno 2] Filen eller katalogen finns inte: './tof_list'.`  (File or folder does not exist './tof_list')

I am not sure where it's looking for tof_list, there is a tof_list folder in root so it's a bit weird that I get this error. 

I am running Fedora 32 with the latest version of Python, haven't installed anything outside of the standard repos. I assume this issue is universal to all Linux distro's but I cannot confirm. I tried installing Potku in every reasonable location (including directly in my home directory), but that doesn't seem to work. I have followed the build instructions to the letter, and haven't gotten any issues there.

jussiks, 2020-08-26:
Hi SjoerdB93,

thanks for bringing this up. I think I know what the problem is.

The [Getting started with development](https://github.com/JYU-IBA/potku#getting-started-with-development) section is outdated, specifically the part that says to checkout development branch. Nowadays the latest version is in the master branch.

Development branch has a version of tof_list that does not compile on Linux. Switch to master branch and recompile the 'external' programs.

```
$ git checkout master
$ ./build.sh
```

After running those commands, tof_list executable should be found in `potku/external/bin` folder.

jussiks, 2020-08-31:
I updated the README. Are you still experiencing the issue with tof_list?

SjoerdB93, 2020-09-01:
Hey,

I tried building it again, from the master branch, and indeed the problem with tof_list disappeared. 
However, now I get the following error when trying to create a depth profile:
`2020-09-01 11:44:56 - ERROR - Could not create Depth Profile graph: '>' not supported between instances of 'float' and 'NoneType' `

So I guess there's some data somewhere that does not get stored properly, or at least at some point it's pointing to an empty variable. Note that I just imported a random ERDA measurement and created quick cuts. I did not do any callibration or whatsoever. 

Also, when I tried to start potku, I got an error about the clock module not being found. This is because time is deprecated in Python 3.8, which I am using. A quick fix I found online was replacing the from `time import clock` line with "`from time import perf_counter as clock`". I'm not sure if that leads to other problems either.

I did find that there's a precompiled version that doesn't require external dependencies on the home-page, that one works as intended: https://www.jyu.fi/science/en/physics/research/infrastructures/accelerator-laboratory/pelletron/potku/release_versions

jussiks, 2020-09-02:
> Hey,
>
> I tried building it again, from the master branch, and indeed the problem with tof_list disappeared.
> However, now I get the following error when trying to create a depth profile:
> `2020-09-01 [...]

This error may be caused by a missing `potku/external/share/srim2013.tot` file. The file is not included in the repo, but it comes packaged with the [release version](https://www.jyu.fi/science/en/physics/research/infrastructures/accelerator-laboratory/pelletron/potku/release_versions) from where it can be copied.

> Also, when I tried to start potku, I got an error about the clock module not being found. This is because time is deprecated in Python 3.8, which I am using. A quick fix I found online was replacing [...]

Thanks for the heads up. Potku has been (mostly) developed on 3.6, but it's always better to be compatible with newer versions of Python.

I replaced clock with default_timer from timeit module. It's the exact same thing as perf_counter but default_timer has already been used elsewhere in the source code so using it is more consistent.

> I did find that there's a precompiled version that doesn't require external dependencies on the home-page, that one works as intended: https://www.jyu.fi/science/en/physics/research/infrastructures/accelerator-laboratory/pelletron/potku/release_versions

Based on the date of that Linux version and recent commit history, there are at least a couple of crash fixes that are missing from the release version. It is also missing `coinc`, an external program used when importing .evnt files. See [Readme](https://github.com/JYU-IBA/potku#missing-binaries) and #61 for more info.

Running `python run_potku.py` in an up-to-date master branch is currently the best way to run Potku on Linux.

SjoerdB93, 2020-09-02:
The srim2013.tot file was missing indeed, I took it from the release version and copied it to my Potku folder after building it from the master branch. I initially got the same error (`not supported between instances of 'float' and 'NoneType'`) Then I decided to do the depth profiling a bit more properly and added the efficiency files and did a calibration. That seems to have solved the issue, the depth profiling works exactly as intended!

Maybe it needs the efficiency files in order to do depth profiling? It could explain why it has a NoneType variable as it may wanted to get some variable from the efficiency files? (Needless to say, I generally always use the efficiency files as provided by Uppsala University, but I didn't import them to Potku last time when I just wanted to see if depth profiling works in the first place)

The release version indeed has some crash issues. At least when I go to the graph setting (right-click on the ToF histogram and then click graph settings), it crashes immediately. I worked around this by going to the request settings instead. (I needed to change the compression on both axes as I find the default compression a bit excessive)

jaakkojulin, 2020-09-04:
Efficiency files are not necessary, calibration probably is (at least in practice). Somebody could check if this error reported by @SjoerdB93 has anything to do with that.

Nevertheless since depth profiling seems to work after instructions were revised, support for Python 3.8 was improved (thanks @jussiks) and srim2013.tot (see issue #66) was copied in place I'm closing the issue.

Please submit a new issue if you experience (excessive) crashes with the GitHub version!

closed 2020-09-04 by jaakkojulin