JYU-IBA/potku #79: potku2.ini color_scheme KeyError

tutapitk, 2020-09-13:
# Description

Opening a measurement with an older version of Potku (after using a relatively new version of Potku) causes the following error:
```
Traceback (most recent call last):
  File "potku.py", line 605, in focus_selected_tab
  File "widgets\measurement\tab.py", line 139, in add_histogram
  File "widgets\measurement\tofe_histogram.py", line 61, in __init__
  File "widgets\matplotlib\measurement\tofe_histogram.py", line 133, in __init__
  File "widgets\matplotlib\measurement\tofe_histogram.py", line 212, in on_draw
KeyError: 'jet'
```

# Cause

Old `~/potku/potku2.ini` (e.g. 2019-05-28):
```
[tof-e_graph]
...
color_scheme = Default color
```

New `~/potku/potku2.ini` (e.g. 2020-09-04):
```
[tof-e_graph]
...
color_scheme = jet
```

# Workaround

After changing Potku version:
```
Global Settings... > OK
```
This will update the potku2.ini file.

jussiks, 2020-09-14:
Oh darn, did not notice that the old version serializes the human-readable color scheme value instead of the one that gets passed down to matplotlib...

Potential fix: add from_string method to modules.enums.ToFEColorScheme
```python
_TOFE_MAPPING = {
    "Default color": "jet",
    "Greyscale": "Greys",
    "Greyscale (inverted)": "gray",
}

@enum.unique
class ToFEColorScheme(str, Enum):
    DEFAULT = "jet"
    GREYSCALE = "Greys"
    INV_GREYSCALE = "gray"

    def __str__(self):
        if self is ToFEColorScheme.DEFAULT:
            return "Default color"
        if self is ToFEColorScheme.GREYSCALE:
            return "Greyscale"
        return "Greyscale (inverted)"

    @classmethod
    def from_string(cls, string: str):
        return cls(_TOFE_MAPPING.get(string, string))
``` 

And then make GlobalSettings save the string representation:
```python
    def set_tofe_color(self, value: ToFEColorScheme):
        """Set  color of the ToF-E Histogram.

        Args:
            value: a ToFEColorScheme value.
        """
        self.__config[self._TOFE]["color_scheme"] = str(value)
```

Should work with both new and old `.ini`-files

jussiks, 2020-09-23:
Went a little back and forth on how to fix this (or whether to fix this at all). I ended up making color scheme deserialization more flexible by adding a `from_string` method (42a5e9f). This seemed like a handy feature so I merged it to master already.

After that I added #81, which changes the way the color scheme is written to `potku2.ini` back to the original format and causes no KeyErrors in older versions. The PR introduces a minor drawback for Potku versions between d0fe3a7 (roughly, not sure of the exact change) and 1b5f398, so not sure if its worth accepting.

PR #81 (2020-09-27): Revert color scheme serialization back to the original way (#79)
Effects of this PR:
- Older (before d0fe3a7) versions of Potku no longer crash when trying to use .ini files that were produced by newer (1b5f398 and after) versions of Potku.
- Versions between d0fe3a7 and 1b5f398 cannot read color schemes from .ini files made by older or newer versions of Potku, but will revert to default value without crashing.

Closes #79.
closed 2020-09-27 by jussiks