JYU-IBA/potku #80: Measurement and simulation settings are not copied from request settings

tutapitk, 2020-09-13:
# Description

Default measurement settings do not reflect request settings. New measurements (and simulations) do not copy request settings, instead they draw their settings from other default objects.

The picture below illustrates the issue. `Name`, `Beam ion` and `Isotope` should be the same for both, but they are not.

[image]

Previously, Potku copied settings settings from request to newly created measurements and simulations (I think).

# Affected versions

All version from commit fc2d4bc4358f885180561267c95f798c541339cf onwards

jussiks, 2020-09-28:
I suggest doing this via `AdjustableSettings` interface instead of using deepcopy. The idea behind this interface is that you should be able to get and set all attribute values of an object with a single call, which makes it easier to populate GUI with the right values. It also works just fine for copying values from one object to another.

Basically, you'd do either:
```python
self.run = Run(**self.request.default_run.get_settings())
``` 
or
```python
self.run = Run()
self.run.set_settings(**self.request.default_run.get_settings())
```

You will probably have to improve and tweak the implementation of this interface in some classes to suit your needs as well as make some changes in GUI code. But given that `AdjustableSettings` was meant for tackling problems just like this, I think it's worth giving it a go.

tutapitk, 2020-09-28:
`AdjustableSettings` makes sense. I'll use that, thanks.

tutapitk, 2020-10-04:
I tried it out and I'm not sure if `AdjustableSettings` is the best solution for this. `get_settings` returns settings that can be adjusted, but not those that are set just once. There are several places where this an issue. For example, `<measurement>.request.default_run.get_settings()` does not return `Beam`. A copy needs to be created manually. `Detector.foils` is a similar issue.

Also, some of the returned values are objects, so they still have to be deepcopied. This isn't really an issue, but more of an observation that even with `AdjustableSettings`, `deepcopy` is still needed.

The returned member variable names don't necessarily match their constructor names either. For example, `Run` uses `run_time` instead of `time` to prevent shadowing the `time` module. This is problematic in two ways:
- initializing a copy with `kwargs` won't work, because the names won't match
- changing the member variable name will break deserialization

`import time as time_module` is probably the best option there, assuming that `AdjustableSettings` is used.

I think my original approach was a better choice (though still not that great). `AdjustableSettings` has too many gotchas for this purpose. Maybe `deepcopy` with some manual adjustments is good enough.

tutapitk, 2020-10-04:
Oh, I forgot to mention in the commit message that I didn't use `AdjustableSettings` in `Simulation`.

tutapitk, 2020-10-04:
Creating and inheriting a class like `ExportableSettings` (`export_settings` -> a copy of all settings needed for initialization) or `Copiable` (`copy` -> a deepcopy) is also an option.

jussiks, 2020-10-06:
Even if values returned by `AdjustableSettings.get_settings` cannot be used as kwargs for `__init__`, they should still be usable as kwargs for `AdjustableSettings.set_settings` method. This is something that should be enforced in all `AdjustableSettings` implementations, as it is an important feature of the interface. 

For example this should work as intended:

```python
# in Measurement.__init__
if run is None:
    self.run = Run()
    self.run.set_settings(**self.request.default_run.get_settings())
    self.run.beam.set_settings(
        **self.request.default_run.beam.get_settings())
else:
    self.run = run
```

As long as values returned by `get_settings` are non-mutable, there is no need for deepcopying (non-mutability is another thing that would be good to enforce).

Detector foils (and timing foils) are a problem since they they are both mutable and they are stored in a mutable list. Values could still be copied by using the `Foil.generate_foil` factory method.

```python
copied_foils = [
    Foil.generate_foil(**foil.to_dict())
    for foil in self.request.default_detector.foils
]
```

For this to work, `Foil.generate_foil` also be able to initialize Layer objects which it currently does not do.

`AdjustableSettings` could be made more flexible by making the base class handle nested instances of AdjustableSettings automatically. So something like this:

```python
class AdjustableSettings:
    def _get_attrs(self) -> Set[str]:
        # if settings were properties, this could be done using reflection
        raise NotImplementedError

    def _get_setting_value(self, attr) -> Any:
        value = getattr(self, attr)
        if isinstance(value, AdjustableSettings):
            return value.get_settings()
        return value

    def _set_setting_value(self, attr, value):
        attr_val = getattr(self, attr)
        if isinstance(attr_val, AdjustableSettings):
            attr_val.set_settings(**value)
        else:
            setattr(self, attr, value)

    def get_settings(self) -> Dict[str, Any]:
        return {
            attr: self._get_setting_value(attr)
            for attr in self._get_attrs()
        }

    def set_settings(self, **kwargs):
        allowed_attrs = self._get_attrs()
        for k, v in kwargs.items():
            if k in allowed_attrs:
                self._set_setting_value(k, v)
```

jussiks, 2020-10-08:
I did forget that there are `Element` objects whose values need to be copied too. Ideally `Element`s would be immutable, but the `amount` attribute complicates things. However, you can still use a combination of `Element.from_string` - `Element.__str__` to move the values around.

I believe that using and improving Potku's already existing mechanisms makes the code base more robust and easier to understand and reason about. Adding in new mechanisms like deepcopying can be done if they provide a clear improvement, but I prefer making existing code more reusable.

tutapitk, 2020-10-09:
`setSettings` instead of passing parameters to seems good.

Using strings to copy `Element`s is a bit questionable, but I guess it's more native than external methods like `deepcopy`.

I agree that keeping the codebase small is a top priority.

If `AdjustableSettings` automatically gets and sets nested instances of `AdjustableSettings`, it doesn't deal with just immutable objects anymore. I'd say it's still ok because mutable objects are settings too, so they should be adjustable.

jussiks, 2020-10-09:
> Using strings to copy `Element`s is a bit questionable, but I guess it's more native than external methods like `deepcopy`.

True, especially as `__str__` is not generally guaranteed to return a value that can be converted back to the original object.

There is also the `vars` option:
```python
e1 = Element.from_string("1H")
e2 = Element(**vars(e1))
```
or I guess it may not be too much of a hassle to just use element attributes directly:
```python
e1 = Element.from_string("1H")
e2 = Element(e1.symbol, e1.isotope, e1.amount)
```
> 
> If `AdjustableSettings` automatically gets and sets nested instances of `AdjustableSettings`, it doesn't deal with just immutable objects anymore. I'd say it's still ok because mutable objects are settings too, so they should be adjustable.

In my example code above, `get_settings` returns a dictionary whose values are either immutable objects or dictionaries that get mapped to other `AdjustableSettings` objects. So while the nested dictionaries themselves are mutable, you would not be able to set an `AdjustableSettings` object by calling `set_settings`, you can only set its settings.

We are of course getting a bit ahead of ourselves here, because currently there is no type checking or input validation for these settings outside `__init__` methods or GUI elements. #71 could provide some progress on that front.

closed 2020-11-22 by tutapitk