JYU-IBA/potku #86: Use asserts to fail early

tutapitk, 2020-10-11:
The `assert` statement is a tool for making sure that assumptions about code are correct.

Use cases:
- ensure that an object/attribute exists
- ensure that two objects are separate

Example:
```python
# measurement.py
assert id(self.request.default_run.beam.ion) != id(self.run.beam.ion), "Ion should be a copy"
```

Note that all `assert` statements are removed when running with optimization on (`-O`) [[1](https://docs.python.org/3/reference/simple_stmts.html#assert)]. This *probably* means that bundles created with PyInstaller don't retain `assert` statements either [[2](https://pyinstaller.readthedocs.io/en/stable/usage.html#running-pyinstaller-with-python-optimizations)]. This means that `assert`s don't affect release versions.

jussiks, 2020-10-11:
Potku tends to use overly broad exception clauses (i.e. `except Exception: pass`) which means you may not get very helpful errors out of assertions during runtime.

Unit tests can cover most use cases without cluttering the actual code base.  `assert` statements can be useful for quick debugging, but I would not necessarily leave them in the code.

> Note that all `assert` statements are removed when running with optimization on (`-O`) [[1](https://docs.python.org/3/reference/simple_stmts.html#assert)]. This _probably_ means that bundles created [...]

Neither of the two `create_bundle` scripts use `-O` or `-OO` so the bundles are unoptimized. Perhaps this is something to look into. At least file sizes would go down by some amount if the docstrings are dropped.

tutapitk, 2020-10-12:
> Neither of the two `create_bundle` scripts use `-O` or `-OO` so the bundles are unoptimized. Perhaps this is something to look into. At least file sizes would go down by some amount if the docstrings are dropped.

Agreed. Automatic optimizations shouldn't cause any bugs either, so the process is likely simple. If something's wrong, it's likely to be easily noticeable. I wouldn't expect any speed improvements though, the heavy calculations are done with external programs.

I'd argue that asserts can still have benefits. Compared to unit tests, they display the developer's intensions more succintly, without mock instantions or test functions. They are are also in-line, which makes learning the codebase faster. In a way, they are lightweight unit tests.

The assert checks will also be constantly run while developing in debugging mode. This means that they can find errors that don't occur in mocked unit tests. Sadly, it would also mean that running partially refactored code gets harder.

I'm not sure if broad exception clauses would hamper the use of assertions. I envision that the asserts are placed before exceptions get raised.

It's possible that the asserts will just clutter up the codebase. They also need time to write, which could be better spent developing new unit tests or cleaning up code otherwise. Wrong asserts (based on buggy behavior or misunderstandings) are also a risk, though this affects all other parts of programming too.

Still, trying asserts out with new code doesn't take too much effort and removing them afterwards is easy. Therefore, they are worth a try. I probably won't be actively developing Potku for long enough to try them myself, so I'll let future developers choose.

closed 2021-08-31 by tpitkanen