JYU-IBA/potku #108: Default compression level too high

tutapitk, 2020-12-08:
The default compression level (6) is too high for a typical 1920x1080 monitor. For example, on Windows 10 there's roughly 780 vertical pixels for the histogram. In this case, a compression of 10 is enough to achieve a roughly native resolution.

| Compression | Y pixels | Total pixels |
|------------:|---------:|-------------:|
| 10          | 800      | 640k         |
| 6           | 1333     | 1778k        |
| 5           | 1600     | 2560k        |

If I understand the code in `widgets/matplotlib/measurement/tofe_histogram.py` correctly, the resolution for a histogram is pretty much just `x/x_compression` and `y/y_compression`, where `x` and `y` are 8000.

Default settings should work out-of-the-box for all users, not just for those with powerful computers. Therefore, I think the compression level should be set back to 10 instead of 6. This drops the number of virtual pixels to roughly a third, which should improve performance a lot.

Default compression levels were last changed in da7160112dfb7d8e9922e23fc5d82bdc87b0a5ce (5 -> 6) and 20d7f00e8a3a373176720a19f668a15c4793ca04 (10 -> 5). Both changes were done in 2018.

tutapitk, 2020-12-08:
Users have had issues with the performance of element selection (via email and in issue #76). This change should fix it.

jaakkojulin, 2020-12-09:
Compression can be selected in global options. Number of bins in either axis can be much lower than 8192. In fact in can be just 2048 or 1024 for some users. Typical values of compress range from 2 to 8, so a default of 10 sounds excessive and may yield histograms of 100x100 pixels for some users. I personally have a default of 4 and I often opt to reduce it to 3.

The choice of "automatic compress" should be based maybe by targeting histograms with 1000x1000=1 million pixels. If performance is poor, then the code should be improved...

tutapitk, 2020-12-09:
> If performance is poor, then the code should be improved...

Seems like the entire histogram is redrawn whenever a new selection point is added. Fixing this would probably make compression levels a non-issue. Maybe the selection could be drawn on a separate canvas and added to the main histogram once finished.

I'd argue that if the number of bins is 2048 or 1024, then the compression level needs to be manually dropped anyway. 8192 seems to be a common bin count, and 1920x1080 is the most common resolution, so targetting that combination as a baseline makes the most sense to me.

My change mostly impacts those who don't yet know what the compression levels do, i.e. new users. Those users will probably start by playing around with sample data (which is in 8192x8192). If that is needlessly slow, then their first impression of Potku will be bad.

Apparently, on an early 2015 dual-Core i5 MacBook Pro a selection point appears with a 2-3 second delay (with compression level 6). This is pretty bad for a reasonably recent computer so that's why I set the compression to a pretty big number.

The default compression level can be lowered once the performance is fixed. Until then, I think we should make sure that Potku will run smoothly by default for all users.

jaakkojulin, 2020-12-09:
Performance doesn't matter if the program is impossible to use, as is the case with too high compression. 10 is too high **always**. Users should do something to their data before importing if they need that high compression. Display resolution hardly matters, there is a zoom-tool for a reason.

Changing defaults back and forth is counterproductive. I know 10 is too high, please revert commit 478ef60be799972b8136fdec0198fbce13cc3e34.

tutapitk, 2020-12-09:
Alright, I'll revert it.

closed 2020-12-09 by tutapitk

jussiks, 2020-12-09:
Loading and processing of the data can probably be optimized with numpy, like ePotku did. I suggest creating a `ToFEData` class that handles (de)serialization and data transformations. This would make it easier to try out different implementations.