JYU-IBA/potku #110: Improve element selection speed

tutapitk, 2020-12-09:
Element selection points appear with a delay when making selections. The reason for this is that the entire histogram is redrawn whenever a new selection point is added.

Maybe the selection could be done on a separate canvas and then added to the main histogram once finished. This would make selections more responsive.

Updating existing graphs instead of redrawing them is also worth looking into. I'm not sure if this is possible in Potku's case with Matplotlib. The main issue with this is that lines must be drawn between the selection points, so the graph data can't just be manually updated (probably).

More information in #108.

jussiks, 2020-12-09:
> Element selection points appear with a delay when making selections. The reason for this is that the entire histogram is redrawn whenever a new selection point is added.
> 

Not only that, but the histogram is drawn twice when it is initialized (on lines 128 and 149):
https://github.com/JYU-IBA/potku/blob/3c4b70f453049700ee788f17863b8c542a66d143/widgets/measurement/tab.py#L128-L149

Depth profiles also used to be redraw everything for each single change. Performance was improved by storing plotted depth profiles and limit lines separately and then only updating the ones that needed updating:
```python
# Draw initial plot
plotted_line, = axes.plot(xs, ys)
canvas.draw()

# When line needs updating
plotted_line.set_xdata(new_xs)
plotted_line.set_ydata(new_ys)
canvas.draw()
canvas.flush_events()
```
Something similar may work with selections.

Line charts and limit lines were abstracted to their own classes in https://github.com/JYU-IBA/potku/blob/master/widgets/matplotlib/mpl_utils.py.

tleppala, 2023-09-27:
PR #271 Draws histogram as image, and allows faster selection making as result. It still isn't instantaneous but improvement to original speed.

closed 2023-09-27 by tleppala