JYU-IBA/potku #111: Multiple issues with pipfile

jaakkojulin, 2020-12-14:
- [Pipfile.lock](../tree/master/Pipfile.lock) is in the repository (as is recommended, see [discussion](https://github.com/pypa/pipenv/issues/598)) but it is clearly out-of-date (last commit is 1652b470b7558bac42db7e162ee6606bd72f5ab2)
- Shapely needs a version number minimum for macOS Big Sur. Shapely 1.6.4 didn't work, but everything works with 1.7.1, there were issues with finding dylibs), I therefore recommend adding a version number >= 1.7.1 or updating the lockfile.
- Pipfile.lock is impossible to create on macOS, there are issues with pypiwin32 and pywin32. Apparently the two lines for these are supposed to work also on Linux based on googling the issue, but other people have had issues with this, too.  The error message I get is: ` pipenv.patched.piptools.exceptions.NoCandidateFound: Could not find a version that matches pywin32`. No workaround is necessary, but this prevents me (mainly a mac user with Potku) from fixing these issues myself.

jussiks, 2020-12-14:
[That ](https://github.com/JYU-IBA/potku/commit/bcc70d89b654d0c3a49c84cdd1da560c64d72112) should do it. To be honest, I just always commented out those dependencies instead of actually figuring out how to fix this issue...

jaakkojulin, 2020-12-14:
Well, that is also what I did... the commenting out might make some difference to the resulting lockfile, so I didn't risk it.

jaakkojulin, 2020-12-16:
I think PR #112 solves these issues.

closed 2020-12-16 by jaakkojulin