JYU-IBA/potku #113: Custom tools in depth profile window are updating one click behind

aleekaup2, 2021-02-01:
### Reproduction

For example when using the Limit setting tool
1. Select limit setting tool
2. Select lower and upper limits -> concentrations are correct
3. Select new lower and upper limits -> concentrations are wrong
4. Select other limit again -> concentrations update to correct value

Also the changes made in the "graph's element settings" dialog are not visible in the graph until another click

jussiks, 2021-02-01:
5befd96 should fix the issue with legend box. Depth profile widget uses `draw_and_flush` decorator to update the legend after setting new values. Because the widget is not a GraphWrapper instance,  `getattr(canvas_wrapper, "do_flush", False)` always return false and so the legend was not updated until later.
This was a rather quick fix so I am not sure if there is still need to fix something else.

jussiks, 2021-02-01:
Could you check #117 and see if the depth profile works okay?

aleekaup2, 2021-02-02:
#117 fixes this

closed 2021-02-02 by aleekaup2