JYU-IBA/potku #132: Detector settings are not loading

aleekaup2, 2021-03-18:
### Feedback from potku users:

When creating a new request, Potku generates default settings that are partly hard-coded in "detector.py". So one always has to start with a Cl beam and a 4-foil detector.

The beam and geometry settings can easily be changed and stored once and reloaded in future requests. However, we did not manage to load detector settings. They can be stored, look great, but when you load them Potku reliably and silently crashes ("detector.py, line 248, 'str' object has no attribute 'open'").

We tried it on 3 different PCs, I guess that this is a programming bug. One can solve the problem by copying setting files by hand after creating the request and restarting Potku.

In a later version of Potku the program could check if there is an external setting file and, if so, replace the hard-coded values. This already seems to be prepared in the user/potku/presets directory and was maybe the original idea.

jaakkojulin, 2021-03-18:
The issue is probably here: https://github.com/JYU-IBA/potku/blob/057e727575afd121cc1b46727a7ab613ec4d47d4/widgets/detector_settings.py#L185

The `file` is a `'str'` object and not a `Path`. Possible fix might be something like this.

```diff
-        temp_detector = Detector.from_file(file, self.request, False)
+        temp_detector = Detector.from_file(Path(file), self.request, False)
```

tutapitk, 2021-03-18:
The conversion to `Path` object should probably be done in the `Detector.from_file(...)` method like in `Detector.__init__(...)`. This way the issue won't reoccur if `Detector.from_file(...)` is called elsewhere.

jussiks, 2021-03-18:
> The conversion to `Path` object should probably be done in the `Detector.from_file(...)` method like in `Detector.__init__(...)`.

Disagree with this one. Conversion from string to Path should be done as soon as possible and backend API should be strict about using the correct type. Type hints, static analysis and unit tests are the preferred way to prevent these kind of bugs.

tutapitk, 2021-03-18:
> > The conversion to `Path` object should probably be done in the `Detector.from_file(...)` method like in `Detector.__init__(...)`.
> 
> Disagree with this one. Conversion from string to Path should be done as soon as possible and backend API should be strict about using the correct type.

That's probably the best course of action, as long as it's applied consistently. Currently the same conversion is done (in `__init__` and possibly elsewhere) for at least Request, Simulation, Measurement and Detector despite them having the `Path` type hint.

I think the best fix for this issue would to change the following to return a `Path` (though I'm not working on Potku so feel free fix whatever way you see fit):
https://github.com/JYU-IBA/potku/blob/057e727575afd121cc1b46727a7ab613ec4d47d4/widgets/detector_settings.py#L179

> Type hints, static analysis and unit tests are the preferred way to prevent these kind of bugs.

Agreed. Type hints for `fdialogs` would have caught this, though they aren't completely reliable. See 
https://github.com/JYU-IBA/potku/blob/057e727575afd121cc1b46727a7ab613ec4d47d4/widgets/detector_settings.py#L186
for the nearest example ("Unresolved attribute reference 'get_settings' for class 'Serializable'").

jussiks, 2021-03-19:
> Agreed. Type hints for `fdialogs` would have caught this, though they aren't completely reliable. See [...]

`Detector.from_file` has no type hint for its return value so it defaults to `Serializable`. This can be fixed by either adding the type hint or better yet, changing the base class type hint into a generic type:
 
```python
from typing import TypeVar

T = TypeVar("T")

...

class Serializable:
    """Base class for objects that can be serialized into a file and
    deserialized back.
    """
    __slots__ = ()

    @classmethod
    def from_file(cls: T, *args, **kwargs) -> T:
        """Deserializes an object from file.
        """
        raise NotImplementedError

    ...
```

jussiks, 2021-03-19:
> In a later version of Potku the program could check if there is an external setting file and, if so, replace the hard-coded values. This already seems to be prepared in the user/potku/presets directory and was maybe the original idea.

Seems like a good idea. You could open another issue for this.

There is no straightforward way to implement something like that though. Presets were only intended to be handled by settings dialogs so there is no mechanism to access them when creating a new request. At the moment presets can be used for porting settings from one request to another, this just requires more clicks than loading settings automatically from a file.

jussiks, 2021-04-25:
Closing this as the bug was fixed in #133. Feel free to open new issues for further development ideas regarding detector settings.

closed 2021-04-25 by jussiks