JYU-IBA/potku #198: Update Python version

tpitkanen, 2021-09-22:
Potku has been developed for Python 3.6, while Python 3.10 will be released on 2021-10-04. Potku's Python version should be updated at some point to get new Python features, such as:
- [@dataclass](https://docs.python.org/3.7/library/dataclasses.html) for simpler object initialization (Python 3.7)
- [postponed evaluation of annotations](https://docs.python.org/3.7/whatsnew/3.7.html#pep-563-postponed-evaluation-of-annotations) (Python 3.7 with `from __future__ import annotations`, Python 3.10 by default)
- [better error messages](https://docs.python.org/3.10/whatsnew/3.10.html#better-error-messages) (Python 3.10)
- [match-case](https://docs.python.org/3.10/whatsnew/3.10.html#pep-634-structural-pattern-matching) (Python 3.10)
- [zip(strict=True)](https://docs.python.org/3.10/library/functions.html#zip) (Python 3.10)
- support for `__slots__` in classes that inherit from `ABC` (unknown version)
- general bug fixes and performance improvements from later Python versions

Updating the Python version will require some code changes: at least on Python 3.8, some unit tests related to MCERD fail. Libraries (NumPy, Matplotlib etc.) should be updated too.

Also, the latest available Windows installer version of Python 3.6 is 3.6.8, while the newest release is 3.6.14.

jaakkojulin, 2022-04-28:
This is now quite important in my opinion, e.g. installing Python 3.6.15 or older versions via pyenv apparently does not work in latest versions of macOS, at least not without applying some patches.

samivout, 2023-07-14:
Closing outdated issue, Python has been upgraded to 3.10.

closed 2023-07-14 by samivout