JYU-IBA/potku #208: Add support for gas detector

tpitkanen, 2021-10-13:
# Change

(Requested by @mlaitin)

Currently, the only supported detector type is time-of-flight (TOF), as seen in `Request Settings... > Detector > Type:`.

Gas detector support should be added to accommodate ERD/RBS systems that use it instead of TOF.

## Affected code

The entire `Detector` class is handled as TOF, so anything that deals with it is probably affected. The class should be split into `TofDetector` and `GasDetector`, with a common base class for both.

The UI for detector also requires changes, at least `Request Settings... > Detector > Structure` and `Request Settings... > Detector > Calibration` are TOF-specific. (Does a gas detector need software calibration like TOF does?)

In addition, the main *ToF-E Histogram* window for *measurements* needs to be replaced with something more fitting for a gas detector (if selected). The current *Time of Flight (Ch)* and *Energy (Ch)* axes are not suitable.

## Backwards-compatibility

The detector is saved to a separate `Detector` folder, which contains at least a `.detector` file. The file contains TOF-specific settings such as `tof_slope`, `tof_offset`, `tof_foils` (and `timeres`?). Once the new gas detector type is added, the settings that are present in a `.detector` file will change if the new type is selected. 

Potku closes if it encounters unexpected keys or values in settings file, which means that to preserve maximum backwards-compatibility, the settings file for TOF detector shouldn't be changed. Gas type will be incompatible anyway.

tpitkanen, 2021-10-13:
Changes to **simulation** settings (compared to TOF detector):
- Replace **timing foils** with some kind of a **detector foil**. Normal foils are unaffected.
    - Figure out a good **name** for the detector foil.
- Disable **time resolution (s)** setting and enable/add **energy resolution (keV)** (gray out one of the settings).
    - Element-specific energy resolution would be nice (generally better resolution for heavier elements), but it is complicated to implement.
    - Typical value is around 15 - 20 keV.
- Efficiencies are unaffected.
- (Simulations don't need calibration because they use already-calibrated numbers, unlike measurements which are based on detected values.)

jaakkojulin, 2022-04-28:
I now believe MCERD does not require any modifications to support energy detector, if we assume energy detector is whatever is beyond the last foil. The energy after last foil (in MeVs) is output by MCERD as the 4th column in erd-files. The data can be read by `get_espe` also without any changes, just passing `--eres` instead of `--timeres` should be enough. The energy resolution should be given as keVs FWHM.

In the future we might have to change the default behaviour, but let's see. I have made minor changes to MCERD in commit https://github.com/JYU-IBA/mcerd/commit/6fcc6f6aa0a60c199db68a0caadbb66033f70e4a and Potku should be updated to use the latest version. And tested. JIBAL needs to be updated too.

tleppala, 2023-09-27:
Energy detector option is added in PR #274

closed 2023-09-27 by tleppala