JYU-IBA/potku #230: Plot / sum spectrum is not updating when a recoil element is removed

jkpnen, 2021-11-25:
When a recoil element (such as Cl) is removed from the GUI it should be removed from the plot too.

[image]
[image]

jkpnen, 2021-11-25:
@tpitkanen , any ideas how to tackle this?

I think here should be some kind of listener:

https://github.com/JYU-IBA/potku/blob/6d177f01ae153cf6e2288276a4d304f731483049/widgets/matplotlib/simulation/recoil_atom_distribution.py#L363-L373

Is it possible to somehow use this:
https://github.com/JYU-IBA/potku/blob/bafe93071c5b5390cf2347f2457fb0b3b428eb6d/widgets/matplotlib/measurement/energy_spectrum.py#L788-L810

tpitkanen, 2021-11-25:
The "Remove selected element" button calls this method: `RecoilAtomDistributionWidget.remove_current_element`. Try looking into that. Looks like the method should remove the element from plots, but doesn't for some reason.

I think you should first ask @mlaitin if Potku has previously done this or not. If it has, this current behavior is caused by a bug and needs to be fixed. If it hasn't, then this is new functionality that needs to be implemented.

Also, as always, check if this change has any possible side effects.

PR #232 (2022-01-11): Fix sum spectrum issues
- [x] Fix #231 
- [ ] Fix #230 
- [ ] Fix #222