JYU-IBA/potku #250: Efficiency plot is not working

Huojuu, 2022-02-01:
When trying to plot efficiency plots for hydrogen in the detector settings the "Plot efficiencies" button does nothing. (Request settings -> Detector -> Efficiencies -> Plot efficiencies)

The code does read the data from the correct file and goes through the plotting process like any other but does not produce a separate window with the plot like its supposed to.

jussiks, 2022-02-04:
Thanks for reporting the issue. It seems efficiency plotting is not working properly at all:
- [EfficiencyDialog](https://github.com/JYU-IBA/potku/blob/master/widgets/eff_plot.py#L39) does not show up because it was changed from `QWidget` to `QDialog`.
- Wrong files are being plotted. Potku should plot the files that are in `<request>/Default/Detector/Efficiency_files` folder but instead it tries to plot files from the last folder the user used when they added new efficiency files. Not only are the files wrong but this can easily cause crashes because the is no guarantee that the files actually exist in that folder. This code:
https://github.com/JYU-IBA/potku/blob/f7c499ba6a7fb215a3ff216d15e11eb272b4bdea/widgets/detector_settings.py#L630-L643
should be changed to something like this:
```python
    def __plot_efficiency(self):
        """
        Open efficiency plot widget
        """
        eff_files = self.obj.get_efficiency_files(return_full_paths=True)
        dialog = EfficiencyDialog(eff_files, self)
        dialog.exec_()   # need to make EfficiencyDialog a QDialog so that we can call exec_
```

Huojuu, 2022-02-04:
Thanks for the valuable reply. I too noticed that EfficiencyDialog needs to be changed to QDialog to be able to call exec_. However I did not notice that it tried to plot wrong files. Looks to me that this issue is now solved.

closed 2022-02-04 by Huojuu