JYU-IBA/potku #254: Instruction "python run_python.py" fails to PyQT5 missing, but then pip install is broken

mavavilj, 2022-02-26:
I was following the instructions:

https://github.com/JYU-IBA/potku#getting-started-with-development

and I followed them up to:

`python run_potku.py`

which gives 

`ModuleNotFoundError: No module named 'PyQt5'
`

However, now when trying to get PyQt5 in the environment I've just created:

```
pip install pyqt5
Traceback (most recent call last):
  File "/home/myuser/.local/share/virtualenvs/potku-b0HnYcnS/bin/pip", line 5, in <module>
    from pip._internal.cli.main import main
  File "/home/myuser/.local/share/virtualenvs/potku-b0HnYcnS/lib/python3.9/site-packages/pip/_internal/cli/main.py", line 10, in <module>
    from pip._internal.cli.autocompletion import autocomplete
  File "/home/myuser/.local/share/virtualenvs/potku-b0HnYcnS/lib/python3.9/site-packages/pip/_internal/cli/autocompletion.py", line 9, in <module>
    from pip._internal.cli.main_parser import create_main_parser
  File "/home/myuser/.local/share/virtualenvs/potku-b0HnYcnS/lib/python3.9/site-packages/pip/_internal/cli/main_parser.py", line 7, in <module>
    from pip._internal.cli import cmdoptions
  File "/home/myuser/.local/share/virtualenvs/potku-b0HnYcnS/lib/python3.9/site-packages/pip/_internal/cli/cmdoptions.py", line 24, in <module>
    from pip._internal.exceptions import CommandError
  File "/home/myuser/.local/share/virtualenvs/potku-b0HnYcnS/lib/python3.9/site-packages/pip/_internal/exceptions.py", line 10, in <module>
    from pip._vendor.six import iteritems
ModuleNotFoundError: No module named 'pip._vendor.six'
```

tpitkanen, 2022-03-02:
Thanks for reporting the issue. There are plans to update the Python version from 3.6 (#198), but the version update introduces issues such as this that need to be addressed first.

I think the root cause here is that **Pipfile** locks certain modules' versions, including PyQt5. After updating the modules and fixing a few things (`matplotlib.backends.backend_qt4agg` -> `matplotlib.backends.backend_qt5agg` and adding a cast to float in `GUIReporter.__init__.__update_func()`), I could start Potku on Python 3.10, but at least MCERD didn't work.

samivout, 2023-07-15:
Closing outdated issue. Python has been upgraded to 3.10, libraries locked and MCERD is working.

closed 2023-07-15 by samivout