JYU-IBA/potku #283: Update submodule JIBAL

samivout, 2023-05-03:
Update Potku's submodule JIBAL to its newest version and add support for it in Potku.

samivout, 2023-05-03:
This is functional at least on Windows. Linux and Mac functionality should still be tested, as the commandline argument order for jibaltool might be different for the other operating systems.

jaakkojulin, 2023-05-12:
The implemented order of command line arguments should work on Linux and MacOS. Testing is still strongly encouraged.

It might also make sense to check that the return value of running jibaltool (`Popen.returncode`? or `subprocess.run(..., check=True)` is zero (success) or use some other check to see that jibaltool runs correctly, unless the output parsing already handles this, in which case feel free to ignore this suggestion.

closed 2024-08-09 by jaakkojulin